homieiot/homie-esp8266 #731: compilation failure on ESP32

breezer15, 2022-07-14:
Hello,

I want to compile a bare minimum Sketch in PlatformIO for a ESP32dev kit.

I get this error message if I try to compile:

.pio/libdeps/esp32dev/Homie/src/Homie/Boot/BootNormal.cpp: In member function 'virtual void HomieInternals::BootNormal::setup()':
.pio/libdeps/esp32dev/Homie/src/Homie/Boot/BootNormal.cpp:63:139: error: 'SYSTEM_EVENT_STA_GOT_IP' is not a member of 'arduino_event_id_t'
   _wifiGotIpHandler = WiFi.onEvent(std::bind(&BootNormal::_onWifiGotIp, this, std::placeholders::_1, std::placeholders::_2), WiFiEvent_t::SYSTEM_EVENT_STA_GOT_IP);
                                                                                                                                           ^~~~~~~~~~~~~~~~~~~~~~~
.pio/libdeps/esp32dev/Homie/src/Homie/Boot/BootNormal.cpp:64:153: error: 'SYSTEM_EVENT_STA_DISCONNECTED' is not a member of 'arduino_event_id_t'
   _wifiDisconnectedHandler = WiFi.onEvent(std::bind(&BootNormal::_onWifiDisconnected, this, std::placeholders::_1, std::placeholders::_2), WiFiEvent_t::SYSTEM_EVENT_STA_DISCONNECTED);
                                                                                                                                                         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
Compiling .pio\build\esp32dev\lib620\Homie\Homie\Utils\DeviceId.cpp.o
.pio/libdeps/esp32dev/Homie/src/Homie/Boot/BootNormal.cpp: In member function 'void HomieInternals::BootNormal::_onWifiDisconnected(arduino_event_id_t, arduino_event_info_t)':
.pio/libdeps/esp32dev/Homie/src/Homie/Boot/BootNormal.cpp:377:81: error: 'union arduino_event_info_t' has no member named 'disconnected'; did you mean 'eth_connected'?
   Interface::get().getLogger() << F("✖ Wi-Fi disconnected, reason: ") << info.disconnected.reason << endl;
                                                                                 ^~~~~~~~~~~~
                                                                                 eth_connected
.pio/libdeps/esp32dev/Homie/src/Homie/Boot/BootNormal.cpp:380:44: error: 'union arduino_event_info_t' has no member named 'disconnected'; did you mean 'eth_connected'?
   Interface::get().event.wifiReason = info.disconnected.reason;
                                            ^~~~~~~~~~~~
                                            eth_connected
Compiling .pio\build\esp32dev\lib620\Homie\Homie\Utils\Helpers.cpp.o
*** [.pio\build\esp32dev\lib620\Homie\Homie\Boot\BootNormal.cpp.o] Error 1

My platform.ini contains this:

`; PlatformIO Project Configuration File
;
;   Build options: build flags, source filter
;   Upload options: custom upload port, speed and extra flags
;   Library options: dependencies, extra library storages
;   Advanced options: extra scripting
;
; Please visit documentation for the other options and examples
; https://docs.platformio.org/page/projectconf.html

[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
lib_deps = 
    https://github.com/homieiot/homie-esp8266.git#develop
    OneWire
    DallasTemperature
    Neotimer

build_flags = -D PIO_FRAMEWORK_ARDUINO_LWIP2_LOW_MEMORY
build_type = debug
monitor_speed = 115200
upload_speed = 460800
`

My Main.cpp contains this:

`#include <Homie.h>

void setup() {
  Serial.begin(115200);
  Serial << endl << endl;

  Homie_setFirmware("bare-minimum", "1.0.0"); // The underscore is not a typo! See Magic bytes
  Homie.setup();
}

void loop() {
  Homie.loop();
}`

luebbe, 2022-07-15:
I see that the ESP32 build also fails on CircleCI, so probably something has changed in the libraries/framework and homie needs to be adapted.

breezer15, 2022-07-15:
I could solve the problem now for the time being by reverting to an older version of the ESP32 support package with this line of code in the platformio.ini file:

platform = espressif32 @ 3

luebbe, 2022-10-06:
Solved by #735

closed 2022-10-06 by luebbe