homieiot/homie-esp8266 #733: Compilation failure on ESP8266

Keiishu, 2022-09-27:
Hi,

I'm unable to compile my sketch on PlatformIO 6.1.4, with ESP8266 4.0.1.

<details>
<summary>Error</summary>

```
In file included from .pio\libdeps\proxsens_server\Homie\src/Homie/Datatypes/../Utils/Helpers.hpp:5,
                 from .pio\libdeps\proxsens_server\Homie\src/Homie/Datatypes/../Utils/Validation.hpp:6,
                 from .pio\libdeps\proxsens_server\Homie\src/Homie/Datatypes/../Config.hpp:13,
                 from .pio\libdeps\proxsens_server\Homie\src/Homie/Datatypes/Interface.hpp:7,
                 from .pio\libdeps\proxsens_server\Homie\src/Homie.hpp:6,
                 from .pio\libdeps\proxsens_server\Homie\src/Homie.h:4,
                 from src\main.cpp:4:
.pio\libdeps\proxsens_server\Homie\src/Homie/Datatypes/../Utils/../../StreamingOperator.hpp: In instantiation of 'Print& operator<<(Print&, T) [with T = HomieInternals::Property*]':
Compiling .pio\build\proxsens_server\lib69d\AsyncMqttClient\AsyncMqttClient\Packets\ConnAckPacket.cpp.o
src\main.cpp:240:51:   required from here
.pio\libdeps\proxsens_server\Homie\src/Homie/Datatypes/../Utils/../../StreamingOperator.hpp:5:15: error: no matching function for call to 'print(HomieInternals::Property*&)'
    5 | { stream.print(arg); return stream; }
      |   ~~~~~~~~~~~~^~~~~
```

`no matching function for call to 'print(HomieInternals::Property*&)'`
`invalid conversion from 'HomieInternals::Property*' to 'long long unsigned int' [-fpermissive]`

</details>

<details>
<summary>Complete Output</summary>

```
Processing proxsens_server (platform: espressif8266; board: esp07s; framework: arduino)
---------------------------------------------------------------------------------------------------------------------------------------------------Verbose mode can be enabled via `-v, --verbose` option
CONFIGURATION: https://docs.platformio.org/page/boards/espressif8266/esp07s.html
PLATFORM: Espressif 8266 (4.0.1) > Espressif Generic ESP8266 ESP-07S
HARDWARE: ESP8266 80MHz, 80KB RAM, 4MB Flash
PACKAGES:
 - framework-arduinoespressif8266 @ 3.30002.0 (3.0.2)
 - tool-esptool @ 1.413.0 (4.13)
 - tool-esptoolpy @ 1.30000.201119 (3.0.0)
 - toolchain-xtensa @ 2.100300.210717 (10.3.0)
LDF: Library Dependency Finder -> https://bit.ly/configure-pio-ldf
LDF Modes: Finder ~ chain, Compatibility ~ soft
Found 42 compatible libraries
Scanning dependencies...
Dependency Graph
|-- ESP8266 and ESP32 OLED driver for SSD1306 displays @ 4.3.0
|   |-- Wire @ 1.0
|-- Homie @ 3.0.1
|   |-- ArduinoJson @ 6.11.5
|   |-- AsyncMqttClient @ 0.8.2
|   |   |-- ESPAsyncTCP @ 1.2.0
|   |-- Bounce2 @ 2.71.0
|   |-- ESP Async WebServer @ 1.2.3
|   |   |-- ESPAsyncTCP @ 1.2.0
|   |   |-- Hash @ 1.0
|   |   |-- ESP8266WiFi @ 1.0
|   |-- Ticker @ 1.0
|   |-- ESP8266WiFi @ 1.0
|   |-- DNSServer @ 1.1.1
|   |   |-- ESP8266WiFi @ 1.0
|   |-- ESP8266HTTPClient @ 1.2
|   |   |-- ESP8266WiFi @ 1.0
|   |-- ESPAsyncTCP @ 1.2.0
|   |-- ESP8266mDNS @ 1.2
|   |   |-- ESP8266WiFi @ 1.0
|-- Wire @ 1.0
Building in release mode
Compiling .pio\build\proxsens_server\src\main.cpp.o
[... 197 lines cut ...]
                 from src\main.cpp:4:
.pio\libdeps\proxsens_server\Homie\src/Homie/Datatypes/../Utils/../../StreamingOperator.hpp:5:16: warning: invalid conversion from 'HomieInternals::Property*' to 'long long unsigned int' [-fpermissive]
    5 | { stream.print(arg); return stream; }
      |                ^~~
      |                |
      |                HomieInternals::Property*
*** [.pio\build\proxsens_server\src\main.cpp.o] Error 1
=========================================================== [FAILED] Took 26.50 seconds ===========================================================
```

</details>

Apparently, the code causing this issue is from `StreamingOperator.hpp`, line 5 :
```
#pragma once

template<class T>
inline Print &operator <<(Print &stream, T arg)
{ stream.print(arg); return stream; }

enum _EndLineCode { endl };

inline Print &operator <<(Print &stream, _EndLineCode arg)
{ stream.println(); return stream; }
```

luebbe, 2022-09-28:
Hmm, I can compile my projects, but I'm using the #develop branch, not #master. Maybe that's the reason?

Keiishu, 2022-09-29:
Thanks for your response. In the end, it was just a line I forgot to delete... My bad.

closed 2022-09-29 by Keiishu