homieiot/homie-esp8266 #736: MQTT reconnect fails after Wifi-disconnect

dobernhardt, 2022-10-26:
When rebooting the Wifi-AP or temporarily shutting down Wifi for any other reason the device does not reconnect to the MQTT broker but gets stuck in an endless reconnection loop
```
✔ Wi-Fi connected, IP: 192.168.100.90
Triggering WIFI_CONNECTED event...
↕ Attempting to connect to MQTT...
↕ Attempting to connect to MQTT...
↕ Attempting to connect to MQTT...
↕ Attempting to connect to MQTT...
↕ Attempting to connect to MQTT...
↕ Attempting to connect to MQTT...
↕ Attempting to connect to MQTT...
↕ Attempting to connect to MQTT...
↕ Attempting to connect to MQTT...
↕ Attempting to connect to MQTT...
↕ Attempting to connect to MQTT...

```

If only the connection to the MQTT broker is lost a reconnect works without any problems.

I'm using latest e0347c08231592672c16c07356127a8f5d539c26 from develop.

dobernhardt, 2022-10-26:
Probably related to https://github.com/homieiot/homie-esp8266/issues/320 ?

luebbe, 2022-11-02:
OK, that's the sledgehammer method :)
Which version of homie-esp are you using? The develop branch is the most up to date and works fine for me. There hasn't been a relase for a long time and the pio library is far behind.

Pierre-33, 2023-05-21:
Hi,
I'm having exactly the same issue with e0347c0 on an esp8266.

After some test, it appears that it's because I use AsyncMqttClient 0.9.0 instead of the 0.8.2 packaged with Homie. Was that your case as well?

Unfortunately my home assistant auto discovery payload seems too big for  0.8.2, so I really need to upgrade it.

dobernhardt, 2023-05-21:
I was indeed using the version 0.9.0 of the AsyncMqttClient library. I have not tested with an older version and I'm still relying on my sledgehammer workaround resetting the device on WIFI_DISCONNECTED.

luebbe, 2023-05-22:
@bertmelis has developed a much more stable mqtt library (https://github.com/bertmelis/espMqttClient). It is long on my todo list to replace asyncmqtt with bert's library in homie.

Pierre-33, 2023-05-22:
> I was indeed using the version 0.9.0 of the AsyncMqttClient library. I have not tested with an older version and I'm still relying on my sledgehammer workaround resetting the device on WIFI_DISCONNECTED.

I use the same kind of tricks but on the WIFI_CONNECTED event as doing the reboot in the WIFI_DISCONNECTED event somehow made the esp reboot again and again untill the Wifi is back.

Pierre-33, 2023-05-22:
> @bertmelis has developed a much more stable mqtt library (https://github.com/bertmelis/espMqttClient). It is long on my todo list to replace asyncmqtt with bert's library in homie.

That sounds like a lot of works. I was under the impression that Homie development has kind of stopped. I was looking for a tiny synchronous MQTT Client that I could use to publish my HomeAssistant Autodiscovery message when the esp boot and then shut it down and let Homie works with 0.8.2. But all I could find it "huge" async mqtt client that requires an update loop and I didn't feel like having two Mqtt client working in parallel even if it's just during setup.

Pierre-33, 2023-05-22:
yes, that's what I do, but Homie Mqtt Client doesn't supporte "big" payload (my payload are actually quite small). That's why I upgrade it to 0.9 at the first place. And now I got this reconnection issue. So I was looking for a solution to rollback to AsyncMqttClient 0.8.2 and use another client for my "big" payload.

labodj, 2026-10-27:
I maintain a [standalone fork](https://github.com/labodj/homie-esp8266/) of this library where I have reworked Wi-Fi/MQTT recovery and replaced AsyncMqttClient with espMqttClient, which was suggested earlier in this thread.

MQTT connection attempts wait for a Wi-Fi IP address, and the main loop reconciles missed connection events and retries stalled attempts. These changes address the recovery path described here, although I have not reproduced the original M5StickPlus case.

The changes are available in [v4.1.0](https://github.com/labodj/homie-esp8266/releases/tag/v4.1.0).
[Compatibility notes](https://labodj.github.io/homie-esp8266/others/fork-differences/) cover two relevant differences: MQTT TLS is unsupported in 4.x, and sketches accessing Homie.getMqttClient() directly need to adapt to the new client API.
Homie 3.0.1 remains the default convention.