homieiot/homie-esp8266 #740: 'HTTPClient::begin' declared with attribute error: obsolete API, use ::begin(WiFiClient, url)

ulises2k, 2023-02-26:
Version Homie 3.0.1
ESP8266 - 3.1.1
Arduino IDE 1.8.9

[image]

homie-esp8266-3.0.1\src\Homie\Boot\BootConfig.cpp: In member function 'void HomieInternals::BootConfig::_proxyHttpRequest(AsyncWebServerRequest*)':

homie-esp8266-3.0.1\src\Homie\Boot\BootConfig.cpp:315:20: error: call to 'HTTPClient::begin' declared with attribute error: obsolete API, use ::begin(WiFiClient, url)

ulises2k, 2023-02-26:
FIX
[image]

```
void BootConfig::_proxyHttpRequest(AsyncWebServerRequest *request) {
  Interface::get().getLogger() << F("Received transparent proxy request") << endl;

  String url = String("http://");
  url.concat(request->host());
  url.concat(request->url());

	**WiFiClient client;**
	
  // send request to destination (as in incoming host header)
  _httpClient.setUserAgent(F("ESP8266-Homie"));
  _httpClient.begin(**client**,url);
...

```

luebbe, 2023-02-27:
This is a duplicate of #711 and it is fixed on the develop branch. I guess there is no official release with this fix

trysten, 2023-03-15:
Indeed: fixed by #717
I did:
```bash
cd ~/Arduino/libraries
git clone https://github.com/homieiot/homie-esp8266.git --depth=1
```
This issue can be closed.