homieiot/homie-esp8266 #743: PlatformIO espressif8266@4.2.0 linker error

trysten, 2023-07-09:
[Newest](https://registry.platformio.org/platforms/platformio/espressif8266?version=4.2.0) PlatformIO platform package will not build homie-esp8266 (on linux) with:
```
Linking .pio/build/nodemcuv2/firmware.elf
.pio/build/nodemcuv2/libFrameworkArduino.a(core_esp8266_postmortem.cpp.o): in function `__wrap_system_restart_local':
core_esp8266_postmortem.cpp:(.text.__wrap_system_restart_local+0x2): dangerous relocation: j: cannot encode: (.text.postmortem_report+0x88)
collect2: error: ld returned 1 exit status
*** [.pio/build/nodemcuv2/firmware.elf] Error 1
```
platformio.ini:
```
[env:nodemcuv2]
; 4.2.0 is newest at time of writing, locking version for clarity
platform = espressif8266@4.2.0 
framework = arduino
board = nodemcuv2
lib_deps = git+https://github.com/homieiot/homie-esp8266.git#develop 
```
platformio pkg list:
```
Resolving nodemcuv2 dependencies...
Platform espressif8266 @ 4.2.0 (required: espressif8266 @ 4.2.0)
├── framework-arduinoespressif8266 @ 3.30102.0 (required: platformio/framework-arduinoespressif8266 @ ~3.30102.0)
├── tool-esptool @ 1.413.0 (required: platformio/tool-esptool @ <2)
├── tool-esptoolpy @ 1.30000.201119 (required: platformio/tool-esptoolpy @ ~1.30000.0)
└── toolchain-xtensa @ 2.100300.220621 (required: platformio/toolchain-xtensa @ ~2.100300.0)

Libraries
└── Homie @ 3.0.1+sha.e0347c0 (required: git+https://github.com/homieiot/homie-esp8266.git#develop)
│   ├── ArduinoJson @ 6.11.5 (required: ArduinoJson @ ~6.11.4)
│   ├── AsyncMqttClient @ 0.8.2 (required: AsyncMqttClient @ ^0.8.0)
│   │   └── ESPAsyncTCP @ 1.2.0 (required: ESPAsyncTCP @ 1.2.0)
│   ├── Bounce2 @ 2.71.0 (required: Bounce2 @ ^2.1.0)
│   └── ESP Async WebServer @ 1.2.3 (required: ESP Async WebServer)
```
Current workaround is downgrading platform to `platform = espressif8266@4.1.0`

I don't know what this error is about. Yet. We need to lock platform version for stable because right now a new install from PlatformIO will not build.

luebbe, 2023-07-11:
Thanks for the workaround. I ran into the same problem, but didn't know how to solve it.

denis-obukhov, 2023-07-14:
+1 The same issue